Azure-Samples/iot-edge-opc-plc #7: Feature request: Support incoming commands

sandervandevelde, 2019-10-15:
[X] feature request

Hello,

we want to test how the OPC-Twin supports sending commands to OPC-UA servers.

Is it possible to add handling commands to the OPC-PLC? Is this possible due to the usage of the standard OPC-UA server from the [UA-.NETStandard](https://github.com/OPCFoundation/UA-.NETStandard)?
We are happy to help. Can you point us in the right direction on how to add this feature?

hansgschossmann, 2020-01-31:
Sorry for the late response. What do you mean by "add handling commands"?
If it is about adding your own OPC methods, just use the existing as a blueprint.

hansgschossmann, 2020-09-21:
If you still need to know how to run opc-twin standalone and use the direct method interface, pls let me know.

hansgschossmann, 2020-09-21:
Just saw your comment on adding your own methods. OPCTwin has a framework, which wires up methods in a C# class as direct methods. You just need to add the rigth attributes to the class containing the methods.

hansgschossmann, 2021-04-15:
@sandervandevelde pls check the implementation of the existing OPC UA methods "ResetStepUp", "StartStepUp" and "StopStepUp" which should show you how to add your own OPC UA methods.
Pls reactivate the issue, if this is not sufficient.

closed 2021-04-15 by hansgschossmann