Azure-Samples/iot-edge-opc-plc #8: Can't read simulated history data

yenicelik, 2019-12-06:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [x] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>

### Any log messages given by the failure
>

### Expected/desired behavior
> 
Is it possible to retrieve simulated historical data?
And is it possible to use the OPC `read_history` function? I tried changing changing all `Historizing = false,` to `Historizing = true,`. Then I try to read-in the values using

```
            myvar = root.get_child(["0:Objects", "2:OpcPlc", "2:Telemetry", "2:RandomSignedInt32"])
            var = myvar # client.get_node("ns=2;s=Input.IEPE11")
            dt_now = datetime.datetime.utcnow()
            endtime = dt_now + datetime.timedelta(seconds=30)
            starttime = dt_now - datetime.timedelta(seconds=0)
            print("Start and end times are: ", endtime, starttime)
            result = var.read_raw_history(starttime, endtime)
``` 

the error message I get it:

```
Traceback (most recent call last):
  File "/path/mocks/client.py", line 55, in <module>
    result = var.read_raw_history(starttime, endtime)
  File "/path/python3.6/site-packages/opcua/common/node.py", line 510, in read_raw_history
    return result.HistoryData.DataValues
AttributeError: 'NoneType' object has no attribute 'DataValues'
```

But this doesn't work. 

Is it possible to determine this somehow (and determine the kind of data to be simulated) using the config file:? 

Is it possible to determine what kind of data is to be generated? 
```
  "NodeList": [
    {
      "NodeId": "07ZC619-01.OP001",
      "Name": "07ZC619-01.OP001",
      "DataType": "Float",
      "ValueRank": -1,
      "AccessLevel": "HistoryRead",
      "UserAccessLevel": 1,
      "Description": ""
    },
```

### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)

### Versions
>

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

hansgschossmann, 2020-01-31:
I have not looked into this. Are you looking to provide historical access for the user generated nodes or for the random nodes or do you not care. Right now i have no plans to add any of those. Happy to take a PR.

luiscantero, 2022-01-14:
If not solved by the above PRs, please reopen.

closed 2022-01-14 by luiscantero