Azure-Samples/iot-edge-opc-plc #11: Microsoft container entry are not the newest release

Swampen, 2020-02-05:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [x] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> `docker pull mcr.microsoft.com/iotedge/opc-plc`

### Any log messages given by the failure
> ` OpcPlc V1.0.0+1 starting up...`

### Expected/desired behavior
> ` OpcPlc V1.1.0 starting up...`

### OS and Version?
> Linux 18.04 LTS 

### Mention any other details that might be useful
The image on mcr.microsoft.com/iotedge/opc-plc should be updated to the newest release
> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

pcbl, 2020-03-02:
Hi! I noticed the same issue on my end. Basically i would like to use the --Nodesfile parameter but it is not available on the lattest docker image. Would be Possible to make thsi version public?

Swampen, 2020-03-02:
@pcbl Until they decide to publish a newer version, you can set up your own container registry in azure, and publish this module there yourself form VSCode or VS. 
It don't seem like the are looking at issues atm..

pcbl, 2020-03-02:
@Swampen , sure thing. Already working on it. It would just be so convenient when it was already there. :-)

luiscantero, 2020-09-20:
PR #23 fixed this

hansgschossmann, 2020-09-21:
There was an issue in the build scripts which is fixed and latest is now the latest version (1.1.6)

closed 2020-09-21 by hansgschossmann