Azure-Samples/iot-edge-opc-plc #12: Improvement: Add Arrays and Complex Types on Simulation

pcbl, 2020-03-11:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [X] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>

### Any log messages given by the failure
>

### Expected/desired behavior
>It would be nice if we have some Complex Types Generation on the Simulation. One that would create an array of Complex Types, another with a Single Instance just as a more "complex" kind of data that could be obtained from the OPC Server, something like:

- ArrayOfStringData : A simple array with Random String values
- ArrayOfFloatData : A simple array with Random Float values
- ArrayOfComplexData : An array from a given Complex Type(A class with at least 2 Attributes...)
- ComplexData : An instance of a Complex Type(A class with at least 2 Attributes...)

### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)

### Versions
>

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

bzumik1, 2020-07-31:
I would also appreciate this.

luiscantero, 2020-11-19:
What we support now are slow and fast nodes of type "UIntArray". And the option --ctb adds a simple boiler with a complex data type.

algattik, 2021-05-19:
This is now supported with the --ref parameter (#51). @cristipogacean close?

closed 2021-05-20 by luiscantero