Azure-Samples/iot-edge-opc-plc #13: Dockerfile and container configuration

jefim, 2020-03-15:
### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [ ] feature request
- [x] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> -

### Any log messages given by the failure
> -

### Expected/desired behavior
> -

### OS and Version?
> Docker

### Versions
> -

### Mention any other details that might be useful
I am looking to use this in some demoes and I could not really find the Dockerfiles for this repository - https://hub.docker.com/_/microsoft-iotedge-opc-plc?tab=description. There are some links there, but all end up in 404 on GitHub.

Can you provide more details on how to use this container (e.g. how to properly provide configuration for it) and maybe the link to source Dockerfile?

PS I can build my own container on top, of course, but just wondering if there was a way to config the existing container.

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

bzumik1, 2020-07-31:
Hi, how can I configure my own node when I am running in docker?

luiscantero, 2020-09-20:
@jefim @bzumik1 please try this:
powershell -command "& '.\tools\scripts\docker-source.ps1' .\src"
docker build -f .\src\bin\publish\Release\linux-x64\Dockerfile.linux-amd64 -t <YOUR ACR>.azurecr.io/iotedge/opc-plc:latest .\src\bin\publish\Release\linux-x64
az acr login --name <YOUR ACR>
docker push <YOUR ACR>.azurecr.io/iotedge/opc-plc:latest

hansgschossmann, 2020-09-21:
The configuration of the existing container is limited to the user nodes file configuration (option --nodefiles). You can also configuration of slow and fast changing nodes via command line. So far there is no way to device complex types or alarms and events.
The readme.md has been updated with documentation to build your own container.

closed 2020-09-21 by hansgschossmann