Azure-Samples/iot-edge-opc-plc #20: Documentation issue on the docker run options

lucarv, 2020-06-17:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [ ] feature request
- [x] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>
just run the docker run as documented

### Any log messages given by the failure
>
[08:12:11 ERR] Command line arguments: --pn=50000 --autoaccept --nospikes --nodips --nopostrend --nonegtrend --nodatavalues --sp --sn=25 --sr=10 --st=uint --fn=5 --fr=1 --ft=uint

### Expected/desired behavior
>
the container image just starts ;)

### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)
ubuntu

### Versions
>
18.04 LTS

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

grossga, 2020-07-21:
Current documentation for running a Linux docker container notes that the following command will start and run a Linux Docker container:
```
docker run --rm -it -p 50000:50000 -p 8080:8080 --name opcplc mcr.microsoft.com/iotedge/opc-plc:latest --pn=50000 --autoaccept --nospikes --nodips --nopostrend --nonegtrend --nodatavalues --sph --sn=25 --sr=10 --st=uint --fn=5 --fr=1 --ft=uint
```
If anything after `--nodatavalues` is actually supplied in the command an `Error in command line options` message appears with the help.  While the following command works to get a container running, it would be helpful if the documentation provided additional assistance in making the additional parameters work:
```
docker run --rm -it -p 50000:50000 -p 8080:8080 --name opcplc mcr.microsoft.com/iotedge/opc-plc:latest --pn=50000 --autoaccept --nospikes --nodips --nopostrend --nonegtrend --nodatavalues
```

hansgschossmann, 2020-07-24:
ok. got it. pls use opc-plc:1.1.6

latest seems not to be updated and i need to check our devops pipeline. sorry about this.

XC-, 2020-09-12:
This took a while from me as well to notice. Though it can be mitigated with locked version, this information should either be updated to the documentation, or release the 1.1.6 as latest (which I think is the proper solution).Slightly better visibility to the versions would also be nice. It would be great to see from the dockerhub page what tags have been released. Yes, there is a link to the version listing, but this is just a matter of user friendliness.

luiscantero, 2020-09-20:
PR #23 fixed this

hansgschossmann, 2020-09-21:
The latest image (1.1.6) is now pushed correctly into MCR and available to pull.

closed 2020-09-21 by hansgschossmann