Azure-Samples/iot-edge-opc-plc #25: Stop server from using any kind of certificate

andowinger, 2020-10-08:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [ ] feature request
- [x ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>
Set EnableUnsecureTransport to true and remove the block defining the secure ServerSecurityPolicy

### Any log messages given by the failure
>
Any client e(.g. UAExpert) notifies you about an untrusted certificate and you have to manually set this to "trust this certificate" to establish connection

### Expected/desired behavior
>
No certificate should be used if is unsecuretransport is used
### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)
Windows 10
### Versions
>
1809
### Mention any other details that might be useful
The documentation does not provide information what I have to do to completely disable any certificate exchange.

This is the log when starting up - as you can see, there is only a "None" Security policy runing, but still a self-signed-certifiacte is created, though I don't want that here ;)

15:09:16 INF] Web server started on port 8080
[15:09:16 INF] opcstacktracemask set to: 0x0
[15:09:16 INF] OPC UA server base address: opc.tcp://287ed39ad887:50000
[15:09:16 INF] Unsecure security policy http://opcfoundation.org/UA/SecurityPolicy#None with mode None added
[15:09:16 WRN] Note: This is a security risk and needs to be disabled for production use
[15:09:16 INF] Trusted Issuer store type is: Directory
[15:09:16 INF] Trusted Issuer Certificate store path is: pki/issuer
[15:09:16 INF] Trusted Peer Certificate store type is: Directory
[15:09:16 INF] Trusted Peer Certificate store path is: pki/trusted
[15:09:16 INF] Rejected certificate store type is: Directory
[15:09:16 INF] Rejected Certificate store path is: pki/rejected
[15:09:16 INF] Rejection of SHA1 signed certificates is disabled
[15:09:16 INF] Minimum certificate key size set to 1024
[15:09:16 INF] Application Certificate store type is: Directory
[15:09:16 INF] Application Certificate store path is: pki/own
[15:09:16 INF] Application Certificate subject name is: OpcPlc
[15:09:16 WRN] WARNING: Automatically accepting certificates. This is a security risk.
[15:09:16 INF] No existing Application certificate found. Create a self-signed Application certificate valid from yesterday for 12 months,
[15:09:16 INF] with a 2048 bit key and 256 bit hash.
[15:09:17 INF] Application certificate with thumbprint 'AEB28F47284A635E2E1BA044243A49B21AD3DC6D' created.
[15:09:17 INF] Application certificate is for ApplicationUri 'urn:OpcPlc:287ed39ad887', ApplicationName 'OpcPlc' and Subject is 'OpcPlc'
[15:09:17 INF] LDS(-ME) registration interval set to 0 ms (0 means no registration)
[15:09:17 INF] Trusted issuer store contains 0 certs
[15:09:17 INF] Trusted issuer store has 0 CRLs.
[15:09:17 INF] Trusted peer store contains 0 certs
[15:09:17 INF] Trusted peer store has 0 CRLs.
[15:09:17 INF] Rejected certificate store contains 0 certs
[15:09:17 INF] Starting server on endpoint opc.tcp://287ed39ad887:50000 ...
[15:09:17 INF] Simulation settings are:
[15:09:17 INF] One simulation phase consists of 50 cycles
[15:09:17 INF] One cycle takes 100 milliseconds
[15:09:17 INF] Spike generation is disabled
[15:09:17 INF] Data generation is disabled
[15:09:17 INF] Anonymous authentication: enabled
[15:09:17 INF] Username/Password authentication: enabled
[15:09:17 INF] Certificate authentication: disabled
[15:09:18 INF] Creating 25 Slow nodes of type: UInt
[15:09:18 INF] Node values will change each 10 sec
[15:09:18 INF] Node values sampling rate is 0 msec
[15:09:18 INF] Creating 5 Fast nodes of type: UInt
[15:09:18 INF] Node values will change each 1 sec
[15:09:18 INF] Node values sampling rate is 0 msec
[15:09:18 INF] OPC UA Server started.
[15:09:18 INF] OPC Publisher configuration file: pn.json
[
  {
    "EndpointUrl": "opc.tcp://287ed39ad887:50000",
    "UseSecurity": false,
    "OpcNodes": [
      { "Id": "ns=2;s=SlowUInt1", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt2", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt3", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt4", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt5", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt6", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt7", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt8", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt9", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt10", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt11", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt12", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt13", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt14", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt15", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt16", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt17", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt18", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt19", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt20", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt21", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt22", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt23", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt24", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=SlowUInt25", "OpcPublishingInterval": 10000 },
      { "Id": "ns=2;s=FastUInt1" },
      { "Id": "ns=2;s=FastUInt2" },
      { "Id": "ns=2;s=FastUInt3" },
      { "Id": "ns=2;s=FastUInt4" },
      { "Id": "ns=2;s=FastUInt5" }
    ]
  }
]
> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

ajithkumars, 2021-03-23:
Wondering the same.

cristipogacean, 2021-03-23:
This is not possible due to OPC UA spec constraints. All OPC UA applications are required to have an application instance certificate except the nano embedded server profile (which is not the case for this server) even if they only support security policy none.

hansgschossmann, 2021-04-15:
@andowinger @ajithkumars : does @cristipogacean's answer gives you enough clarity. in addition to this what @andowinger what do you mean by "disable TLS"?

ajithkumars, 2021-04-15:
@hansgschossmann : I assuming @cristipogacean  has a point so I choose to believe it. (And find a work around if required)