Azure-Samples/iot-edge-opc-plc #32: Regression in mcr.microsoft.com/iotedge/opc-plc:1.2.0

ajhewett, 2020-11-19:
### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ x] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
```
docker run -it --rm -p "50000:50000" mcr.microsoft.com/iotedge/opc-plc:1.2.0 --plchostname=opcplc --showpnjsonph --portnum=50000 --autoaccept --unsecuretransport
```

### Any log messages given by the failure
```
Unable to find image 'mcr.microsoft.com/iotedge/opc-plc:1.2.0' locally
1.2.0: Pulling from iotedge/opc-plc
Digest: sha256:866d6e4b62ad24b4e84988a84d7c173cae24907074854942374f8b6b24173222
Status: Downloaded newer image for mcr.microsoft.com/iotedge/opc-plc:1.2.0
[14:27:40 INF] Current directory is: /app
[14:27:40 INF] Log file is: /app/dbf88900f3e5-plc.log
[14:27:40 INF] Log level is: info
[14:27:40 INF] OpcPlc V1.2.0 starting up...
info: Microsoft.Hosting.Lifetime[0]
      Now listening on: http://[::]:8080
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: /app
[14:27:40 INF] Web server started on port 8080
[14:27:40 INF] opcstacktracemask set to: 0x0
[14:27:40 INF] OPC UA server base address: opc.tcp://opcplc:50000
[14:27:40 INF] Security policy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 with mode SignAndEncrypt added
[14:27:40 INF] Unsecure security policy http://opcfoundation.org/UA/SecurityPolicy#None with mode None added
[14:27:40 WRN] Note: This is a security risk and needs to be disabled for production use
[14:27:40 INF] Trusted Issuer store type is: Directory
[14:27:40 INF] Trusted Issuer Certificate store path is: pki/issuer
[14:27:40 INF] Trusted Peer Certificate store type is: Directory
[14:27:40 INF] Trusted Peer Certificate store path is: pki/trusted
[14:27:40 INF] Rejected certificate store type is: Directory
[14:27:40 INF] Rejected Certificate store path is: pki/rejected
[14:27:40 INF] Rejection of SHA1 signed certificates is disabled
[14:27:40 INF] Minimum certificate key size set to 1024
[14:27:40 INF] Application Certificate store type is: Directory
[14:27:40 INF] Application Certificate store path is: pki/own
[14:27:40 INF] Application Certificate subject name is: OpcPlc
[14:27:40 WRN] WARNING: Automatically accepting certificates. This is a security risk.
[14:27:40 INF] No existing Application certificate found. Create a self-signed Application certificate valid from yesterday for 12 months,
[14:27:40 INF] with a 2048 bit key and 256 bit hash.
[14:27:41 INF] Application certificate with thumbprint '9A9799313BD331342916BF04EC3196111A564A43' created.
[14:27:41 INF] Application certificate is for ApplicationUri 'urn:OpcPlc:opcplc', ApplicationName 'OpcPlc' and Subject is 'OpcPlc'
[14:27:41 INF] LDS(-ME) registration interval set to 0 ms (0 means no registration)
[14:27:41 INF] Trusted issuer store contains 0 certs
[14:27:41 INF] Trusted issuer store has 0 CRLs.
[14:27:41 INF] Trusted peer store contains 0 certs
[14:27:41 INF] Trusted peer store has 0 CRLs.
[14:27:41 INF] Rejected certificate store contains 0 certs
[14:27:41 INF] Starting server on endpoint opc.tcp://opcplc:50000 ...
[14:27:41 INF] Simulation settings are:
[14:27:41 INF] One simulation phase consists of 50 cycles
[14:27:41 INF] One cycle takes 100 milliseconds
[14:27:41 INF] Spike generation is enabled
[14:27:41 INF] Data generation is enabled
[14:27:41 INF] Anonymous authentication: enabled
[14:27:41 INF] Username/Password authentication: enabled
[14:27:41 INF] Certificate authentication: enabled
[14:27:41 INF] OPC UA Server started.
Unhandled exception.Unhandled exception.Unhandled exception.  Unhandled exception.  Unhandled exception. Unhandled exception. [14:27:41 INF] OPC Publisher configuration file: pn.json
[... 83 lines cut ...]
   at System.Threading.TimerQueueTimer.<>c.<.cctor>b__23_0(Object state)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location where exception was thrown ---
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.TimerQueueTimer.CallCallback(Boolean isThreadPool)
   at System.Threading.TimerQueueTimer.Fire(Boolean isThreadPool)
   at System.Threading.TimerQueue.FireNextTimers()
   at System.Threading.TimerQueue.AppDomainTimerCallback(Int32 id)

```

### Expected/desired behavior
The previous version worked:
```
docker run -it --rm -p "50000:50000" mcr.microsoft.com/iotedge/opc-plc:1.1.6 --plchostname=opcplc --showpnjsonph --portnum=50000 --autoaccept --unsecuretransport
```

### OS and Version?
- macOS Catalina Version 10.15.7
- docker desktop community Version 2.5.0.1 (49550)

### Versions
mcr.microsoft.com/iotedge/opc-plc:1.2.0

### Mention any other details that might be useful

luiscantero, 2020-11-19:
@ajhewett please add the parameter `--ctb` for now, we are already deploying a fix.

hansgschossmann, 2020-11-19:
Fixed with https://github.com/Azure-Samples/iot-edge-opc-plc/pull/31, container is now available in MCR

closed 2020-11-19 by hansgschossmann