Azure-Samples/iot-edge-opc-plc #43: Unable to generate Alarms

jaypaddy, 2021-03-29:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [ ] feature request
- [X] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>
sudo docker run --rm -it -p 60000:60000 -p 8080:8080 --name opcplc mcr.microsoft.com/iotedge/opc-plc:latest --pn=60000 --autoaccept --nospikes --nodips --nopostrend --nonegtrend --nodatavalues --sph --sn=25 --sr=5 --st=uint --fn=5 --fr=1 --ft=uint --ctb --alm

### Any log messages given by the failure
>
[16:42:43 INF] Current directory is: /app
[16:42:43 INF] Log file is: /app/471b451f14a4-plc.log
[16:42:43 INF] Log level is: info
[16:42:43 ERR] Error in command line options
**[16:42:43 ERR] Command line arguments: --pn=60000 --autoaccept --nospikes --nodips --nopostrend --nonegtrend --nodatavalues --sph --sn=25 --sr=5 --st=uint --fn=5 --fr=1 --ft=uint --ctb --alm**
[16:42:43 INF] 
[16:42:43 INF] OpcPlc V1.2.2.3
[16:42:43 INF] Informational version: V1.2.2+6f16489859
[16:42:43 INF] 
[16:42:43 INF] Usage: opcplc.exe [<options>]
[16:42:43 INF] 
[16:42:43 INF] OPC UA PLC for different data simulation scenarios
[16:42:43 INF] To exit the application, just press CTRL-C while it is running.
[16:42:43 INF] 
[16:42:43 INF] To specify a list of strings, please use the following format:
[16:42:43 INF] "<string 1>,<string 2>,...,<string n>"
[16:42:43 INF] or if one string contains commas:
[16:42:43 INF] ""<string 1>","<string 2>",...,"<string n>""

### Expected/desired behavior
>Im trying to get boiler and alm. I'm unable to run it with --alm as it is not recognized and it throws the ERR in the stdout. Also with --ctb, i don't see any new nodes published. Might be i don;t understand  --ctb. I expected it to publish nodes for the boiler.

specifying --ctb without --alm the PLC server runs but did not see any nodes for boiler.

### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)
Linux Ubuntu

### Versions
> Ubuntu 18.04

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

buzzfrog, 2021-04-09:
Strange,
My understanding was that --alm is included in the version 1.2.3. The version you are testing is 1.2.2.3.
Let's investigate

buzzfrog, 2021-04-09:
Hi again
I tested the latest version locally with your parameters and it works. 
--autoaccept --nospikes --nodips --nopostrend --nonegtrend --nodatavalues --sph --sn=25 --sr=5 --st=uint --fn=5 --fr=1 --ft=uint --ctb --alm

Maybe this hasn't been built as a container yet?
BTW: When using --ctb you should see new nodes as children of Server: Green and Yellow

hansgschossmann, 2021-04-15:
Sorry about this. I think the reason is that our team has done restructuring of the MCR publishing process and i had no chance yet to adjust opcplc.
Local build should work. I will look into the MCR problem and resolve it as part of this issue.

luiscantero, 2021-04-15:
@jaypaddy this should be working now, please confirm, thanks. CC @buzzfrog

buzzfrog, 2021-04-16:
I tested with the following command:
`docker run --rm -it -p 60000:60000 -p 8080:8080 --name opcplc mcr.microsoft.com/iotedge/opc-plc:latest --pn=60000 --autoaccept --nospikes --nodips --nopostrend --nonegtrend --nodatavalues --sph --sn=25 --sr=5 --st=uint --fn=5 --fr=1 --ft=uint --ctb --alm`

And from my point of view, it works,

closed 2021-04-16 by luiscantero

jaypaddy, 2021-04-19:
Sorry for the delay. I deleted the previous version and ran the command as put in by @buzzfrog its up and running with the --alm option. Thank you.