Azure-Samples/iot-edge-opc-plc #47: Add writeable nodes for all Built-In data types

hansgschossmann, 2021-04-15:
### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [x ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Mention any other details that might be useful

We need nodes for all OPC UA basic datatypes which are read and writeable.

The data types are (more details can be found here: https://reference.opcfoundation.org/v104/Core/docs/Part3/#8.1):
    8.8 Boolean
    8.9 Byte
    8.10 ByteString
    8.11 DateTime
    8.12 Double
    8.13 Duration
    8.14 Enumeration
    8.15 Float
    8.16 Guid
    8.17 SByte
    8.24 Integer
    8.25 Int16
    8.26 Int32
    8.27 Int64
    8.32 String
    8.34 UInteger
    8.35 UInt16
    8.36 UInt32
    8.37 UInt64
    8.38 UtcTime

hansgschossmann, 2021-04-15:
Additional request: We should also provide an array of type integer and a simple complex type which is writeable.

algattik, 2021-05-19:
This is now covered with --ref (#51). Close?

closed 2021-05-20 by luiscantero