Azure-Samples/iot-edge-opc-plc #53: Mechanism to control value range of simulated nodes

koepalex, 2021-05-14:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ ] bug report 
- [x] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce

Define for fast and slow nodes upper and lower levels to define a range. When the node will change the value it should generate **random** value within that range instead just increment.

### Mention any other details that might be useful

peterbryntesson, 2021-05-17:
I suggest the following settings:
* SlowNodeMinValue, SlowNodeMaxValue
* FastNodeMinValue, FastNodeMaxValue
where the default is min/max of the data type used

We probably need two more settings as well to control whether we randomize of not.
* RandomizeSlowNodes, RandomizeFastNodes - where the default is false.

suneetnangia, 2021-05-18:
@peterbryntesson great proposal.
Looking into the code, I suggest we decouple max and min values from random or sequential assignment of values and allow configurable step/increment size for sequential increments as well. This will potentially allow more realistic scenarios e.g. sensor values are often in a certain range (e.g. 0-100 degree C) and increased in a defined increment size (e.g. 1 degree C).

Example-
1. Sequential with Min=1, Max=100, Step Size=1:  value assignment will start from 1 with increments of 1 (step size), when max value is reached, it will start to recycle and assign value 1 again and follow the order.
2. Random with Min=1, Max=100, Step Size=N/A:  value assignment will occur from any value between 1-100.

Default min/max will be defined internally to the min/max of the type used (e.g. double, unit) so if user does not defined their specific min/max, system will assume the defaults.

These settings can be applied _independently_ to slow, fast and ultra fast nodes.

peterbryntesson, 2021-05-18:
@suneetnangia sounds good - make a draft implementation.

koepalex, 2021-05-18:
@suneetnangia  sounds really good, I like the idea

closed 2021-08-30 by luiscantero