Azure-Samples/iot-edge-opc-plc #56: Unit Testing and Code Quality

suneetnangia, 2021-05-18:
### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [x] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Current state of code does not allow unit testing.
> Code quality is not optimal e.g. swallowing exceptions, highly coupled (NodeType in Program class?).
> Readability issues ( a large section of commented code in PlcNodeManager.cs)

### Any log messages given by the failure
> Errors was swallowed at times and not thrown correctly.

### Expected/desired behavior
> Code is unit testable, consistent, decoupled and a bit more readable.

### Mention any other details that might be useful
If we agree to do this, I can look into working on it in phases, gradually reaching the near optimal quality.

algattik, 2021-05-25:
Also of note is the use of static properties for storing configuration and state.

luiscantero, 2021-08-30:
Thanks @suneetnangia and @algattik . I have made many changes over the past few weeks to decouple the code (e.g. by introducing the concept of plugins). Also, CSE has added unit tests to the solution. Feel free to contribute further in phases by creating PRs. I will close this issue for now.

closed 2021-08-30 by luiscantero