Azure-Samples/iot-edge-opc-plc #68: OPC PLC seem to have issues with responding to session keep-alive requests.

karok2m, 2021-06-23:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce

I am seeing a lot of messaged about OPC PLC not responding to session keep-alive requests when running `TelemetryTransformationTests` test suite of OMP Adapter with OPC Publisher `3.0`. Please note that in the test suite there are scenarios where up to `100` concurrent sessions (`10` assets each having `10` nodes with different publishing intervals) may be opened to consume data from OPC PLC. Though this is not the only test where those logs appear.

Here is the excerpt from test logs:

```log
[15:49:16 WRN] Session endpoint: opc.tcp://opcplclarge:50001/ has Status: BadNoCommunication 'Server not responding to keep alive requests.'
[15:49:16 INF] Outstanding requests: 1, Defunct requests: 0
[15:49:16 INF] Good publish requests: 1, KeepAlive interval: 5000
[15:49:16 INF] SessionId: ns=6;i=1981790945
[15:49:16 INF] Session State: True
[15:49:16 INF] Missed Keep-Alives: 1
[15:49:16 WRN] Session endpoint: opc.tcp://opcplclarge:50001/ has Status: BadNoCommunication 'Server not responding to keep alive requests.'
[15:49:16 INF] Outstanding requests: 1, Defunct requests: 0
[15:49:16 INF] Good publish requests: 1, KeepAlive interval: 5000
[15:49:16 INF] SessionId: ns=6;i=1981790721
[15:49:16 INF] Session State: True
[15:49:16 INF] Missed Keep-Alives: 1
[15:49:16 WRN] Session endpoint: opc.tcp://opcplclarge:50001/ has Status: BadNoCommunication 'Server not responding to keep alive requests.'
[15:49:16 INF] Outstanding requests: 1, Defunct requests: 0
[15:49:16 INF] Good publish requests: 1, KeepAlive interval: 5000
[15:49:16 INF] SessionId: ns=6;i=1981790833
[15:49:16 INF] Session State: True
[15:49:16 INF] Missed Keep-Alives: 1
[15:49:16 WRN] Session endpoint: opc.tcp://opcplclarge:50001/ has Status: BadNoCommunication 'Server not responding to keep alive requests.'
[15:49:16 INF] Outstanding requests: 2, Defunct requests: 0
[15:49:16 INF] Good publish requests: 1, KeepAlive interval: 5000
[15:49:16 INF] SessionId: ns=6;i=1981790721
[15:49:16 INF] Session State: True
[15:49:16 INF] Missed Keep-Alives: 2
[15:49:29 WRN] Session endpoint: opc.tcp://opcplclarge:50001/ has Status: BadNoCommunication 'Server not responding to keep alive requests.'
[15:49:29 INF] Outstanding requests: 1, Defunct requests: 0
[15:49:29 INF] Good publish requests: 1, KeepAlive interval: 5000
[15:49:29 INF] SessionId: ns=6;i=1981790889
[15:49:29 INF] Session State: True
[15:49:29 INF] Missed Keep-Alives: 1
[15:49:29 WRN] Session endpoint: opc.tcp://opcplclarge:50001/ has Status: BadNoCommunication 'Server not responding to keep alive requests.'
[15:49:29 INF] Outstanding requests: 1, Defunct requests: 0
[15:49:29 INF] Good publish requests: 1, KeepAlive interval: 5000
[15:49:29 INF] SessionId: ns=6;i=1981791057
[15:49:29 INF] Session State: True
[15:49:29 INF] Missed Keep-Alives: 1
[15:49:29 INF] Session endpoint: opc.tcp://opcplclarge:50001/ got a keep alive after 1 was missed.
[15:49:29 INF] Session endpoint: opc.tcp://opcplclarge:50001/ got a keep alive after 1 was missed.
[15:49:29 INF] Session endpoint: opc.tcp://opcplclarge:50001/ got a keep alive after 2 were missed.
[15:49:29 INF] Session endpoint: opc.tcp://opcplclarge:50001/ got a keep alive after 1 was missed.
[15:49:29 INF] Session endpoint: opc.tcp://opcplclarge:50001/ got a keep alive after 1 was missed.
[15:49:44 WRN] Session endpoint: opc.tcp://opcplclarge:50001/ has Status: BadNoCommunication 'Server not responding to keep alive requests.'
[15:49:44 INF] Outstanding requests: 1, Defunct requests: 0
[15:49:44 INF] Good publish requests: 1, KeepAlive interval: 5000
[15:49:44 INF] SessionId: ns=6;i=1981790889
[15:49:44 INF] Session State: True
[15:49:44 INF] Missed Keep-Alives: 1
[15:49:44 WRN] Session endpoint: opc.tcp://opcplclarge:50001/ has Status: BadNoCommunication 'Server not responding to keep alive requests.'
[15:49:44 INF] Outstanding requests: 1, Defunct requests: 0
[15:49:44 INF] Good publish requests: 1, KeepAlive interval: 5000
[15:49:44 INF] SessionId: ns=6;i=1981791057
[15:49:44 INF] Session State: True
[... 111 lines cut ...]

Linux (ubuntu 18.04)

### Versions

Image: `mcr.microsoft.com/iotedge/opc-plc:latest-debug`

Version:

```
[14:47:14 INF] Certificate 'CN=OPC Publisher, C=US, S=Washington, O=Microsoft, DC=opcpublisher' will be trusted, because of corresponding command line option.
[15:45:57 INF] Certificate 'CN=OPC Publisher, C=US, S=Washington, O=Microsoft, DC=opcpublisher' will be trusted, because of corresponding command line option.
[15:46:48 INF] Current directory is: /app
[15:46:49 INF] Log file is: /app/opcplc-plc.log
[15:46:49 INF] Log level is: info
[15:46:49 INF] OpcPlc V1.2.4 starting up...
[15:46:49 INF] opcstacktracemask set to: 0x0
[15:46:49 INF] OPC UA server base address: opc.tcp://opcplc:50000
[15:46:49 INF] Security policy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 with mode SignAndEncrypt added
```

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

luiscantero, 2021-09-02:
@karok2m is this still happening with the current version?

luiscantero, 2022-01-14:
Please reopen if you see this again and can help us repro.

closed 2022-01-14 by luiscantero