Azure-Samples/iot-edge-opc-plc #144: Need help with certificates and connecting to ACI

ondrelogin, 2022-06-16:
### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [ ] feature request
- [x] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Click the "Deploy to Azure" button from the readme.

### Any log messages given by the failure
> On the OPC Client software
> `Error establishing a connection: BadNotConnected`
> I'm getting lots of traffic on the OPC server, potentially bots, but when connected to the logs from the azure portal here are some related messages from the OPC Server
```
[17:59:51 ERR] OPC: TCPSERVERCHANNEL ForceChannelFault Socket=027CB571, ChannelId=0, TokenId=0, Reason=BadConnectionClosed 'Remote side closed connection'
[17:59:51 ERR] OPC: 06/16/2022 17:59:51.575 TCPSERVERCHANNEL ForceChannelFault Socket=027CB571, ChannelId=0, TokenId=0, Reason=BadConnectionClosed 'Remote side closed connection'
[17:59:51 ERR] OPC: TCPSERVERCHANNEL ForceChannelFault Socket=03FE6417, ChannelId=0, TokenId=0, Reason=BadConnectionClosed 'Remote side closed connection'
[17:59:51 ERR] OPC: 06/16/2022 17:59:51.780 TCPSERVERCHANNEL ForceChannelFault Socket=03FE6417, ChannelId=0, TokenId=0, Reason=BadConnectionClosed 'Remote side closed connection'
```

### Expected/desired behavior
>That I can connect to the OPC Server

### OS and Version?
> Running `cat /etc/issue` on the connect of the aci returns
> `Debian GNU/Linux 11 \n \l`
> The OPC UA Client is running on my windows 10 machine.

### Versions
> The aci Logs show: `OpcPlc V2.3.0 starting up ...`

### Mention any other details that might be useful

I tried my best to find documentation on the readme about how to run this docker image as an _aci_ and manage the certificates but was not able to find any. I apologize if it's in an obvious place, but I could not find any.

As per the directions on the readme, after you run the _Deploy to Azure_ you should be able to connect using
`opc.tcp://<NAME>.<REGION>.azurecontainer.io:50000`

Looking at the container logs I tried this as well as since it said it was `Starting server on endpoint <endpoint-below> ...`
`opc.tcp://aci-<NAME>-<RNG>-plc1.<REGION>.azurecontainer.io:50000`
both didn't work.

I believe the issues is that I need the certificate (der file) from the aci / opc ua server on my client machine so I can connect but I do not know how to get it. Here is some of the log at the end when the certificates are created:
```
[17:54:05 INF] Application store contains 1 certs
[17:54:05 INF] 01: Subject 'CN=OpcPlc' (thumbprint: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx)
[17:54:05 INF] Trusted issuer store contains 0 certs
[17:54:05 INF] Trusted issuer store has 0 CRLs.
[17:54:05 INF] Trusted peer store contains 0 certs
[17:54:05 INF] Trusted peer store has 0 CRLs.
```

Looking at the logs I see there are several entries where it's creating the certificates on the container instance but that doesn't seem to be a good idea. Also, do I need to copy my OPC UA Client certificates to the server as well? I haven't gotten to that phase yet but that might be something that is needed?

I figure I can't connect because I don't have certificates on my OPC UA Client machine and perhaps I need to get certificates to the OPC UA server eventually, but none of that is clear on how to do with aci.

Can I configure the aci with where to get the certificates? is there other ways?

luiscantero, 2022-06-28:
What client are you using to connect to the opcplc? (pls try UaExpert, for example)

You don't need to get the server certificate or upload the one from the client. The client usually shows a warning and you can accept the risk of connecting to a self-signed certificate and connect to it.

You can get the URI to your server from the Azure Portal, this is where you need to connect to:
[image]

ondrelogin, 2022-06-29:
I was using a custom client, but I downloaded UaExpert and tried and I still get an error and can't connect.
I get the option to 

> Please try another configuration or press 'Ignore' to force trying to connect.

Ideally I would like to have a better solution where I can actually use security and connect to an OPC UA server, hosted in a container in more realistic production scenario. Event I can get around these various prompts of getting it to work, I suspect that if I restart the container, then all the connections would fail, because a new Container would start up and a brand new set of certificates would be created and I'd have to go through this headache again.

Is there not a way to run this project, as an ACI, with some level of certificate security?
If so, can I have some help on how to do that?

luiscantero, 2022-06-30:
In the readme we mention the following, have you tried it like this?
```
Note: Make sure that your OPC UA client uses security policy Basic256Sha256 and message security mode Sign & Encrypt to connect.
```
Restarting the container will indeed reset the certificates. You will have to [mount a share](https://docs.microsoft.com/en-us/azure/container-instances/container-instances-volume-azure-files) to be able to persist the certificates. You will then need to modify the deployment template to pass the options related to the certificate storage, just search for "appcert" and "trustedstore" in the readme file.

SatishBoddu-MSFT, 2022-07-15:
Hello @ondrelogin Just checking back, could you please let us know if you need further help in this matter. Please refer  to the above response and do let us know.

luiscantero, 2022-07-18:
Closing due to no response. Feel free to reopen if we can help.

closed 2022-07-18 by luiscantero