Azure-Samples/iot-edge-opc-plc #160: Wrong architecture for image mcr.microsoft.com/iotedge/opc-plc:2.4.0-linux-arm64v8

sylvaneau, 2022-08-05:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> A mac with Silicon processor (M1, M2)
> docker run -it --rm mcr.microsoft.com/iotedge/opc-plc:2.4.0-linux-arm64v8
> or docker run -it --rm mcr.microsoft.com/iotedge/opc-plc:2.4.0

### Any log messages given by the failure
> WARNING: The requested image's platform (linux/arm/v7) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
> exec ./opcplc: no such file or directory

### Expected/desired behavior
> A working opc-plc container :-)

### OS and Version?
> macOS (Monterey) with M1 processor (arm64/v8 architecture)

### Versions
> opc-plc 2.4.0

### Mention any other details that might be useful
if you perform a `docker inspect` on the image you will find that it has been built with the linux/arm/v7 (32 bits) architecture even the image pointed by the 2.4.0-linux-arm64v8 tag.
It seems it is an error in the deployment pipeline because I've managed to simply build the image myself from a Windows PC using the following command (from your documentation) and it works :
```
.\tools\scripts\docker-source.ps1 .\src
docker build -f .\src\bin\publish\Release\linux-arm64\Dockerfile.linux-arm64v8 -t iotedge/opc-plc .\src\bin\publish\Release\linux-x64
```

It would be amazing if can fix it ! :-)

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

closed 2022-08-13 by luiscantero

luiscantero, 2022-08-13:
@sylvaneau this is now fixed. Thanks for reporting it.

sylvaneau, 2022-08-13:
And thanks for fixing it so quickly! ;-)

PR #165 (2022-08-13): Fix build platforms and update nuget
Fix #160.