Azure-Samples/iot-edge-opc-plc #174: Certificate error : unable to load CRL in trust store

VictorRom, 2022-09-07:
### This issue is for a:
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
Generate an openssl CA (key and certificate). Use the following commands to generate the key and the certificate : 

- Key generation : `openssl genrsa -des3 -out myCA.key 2048`
- PEM Certificate generation : `openssl req -x509 -new -nodes -key myCA.key -sha256 -days 1825 -out myCA.pem`
- CRL generation using CA : `openssl ca -gencrl -keyfile myCA.key -cert myCA.pem -out myCA.crl -config crl_openssl.cnf`

The following command can be used to verify the CLR : `openssl crl -in myCA.crl -CAfile myCA.crt` 

Here is the content of the crl_openssl.cnf : 
```
# OpenSSL configuration for CRL generation
#
####################################################################
[ ca ]
default_ca	= CA_default		# The default ca section

####################################################################
[ CA_default ]
database = index.txt
crlnumber = crl_number

default_days	= 365			# how long to certify for
default_crl_days= 30			# how long before next CRL
default_md	= default		# use public key default MD
preserve	= no			# keep passed DN ordering

####################################################################
[ crl_ext ]
# CRL extensions.
# Only issuerAltName and authorityKeyIdentifier make any sense in a CRL.
# issuerAltName=issuer:copy
authorityKeyIdentifier=keyid:always,issuer:always
```

The PEM certificate has to be transformed to DER format using the following command : `openssl x509 -in myCA.pem -out myCA.der -outform DER`

Afterwards place the certificate in the trust store as described below : 

```
├── own
│   ├── certs
│   │   └── OpcPlc [3330615D8D4794D4581DCBE5FF857F2C7BFAECF2].der
│   └── private
│       └── OpcPlc [3330615D8D4794D4581DCBE5FF857F2C7BFAECF2].pfx
└── trusted
    ├── certs
    │   └── myCA.der
    └── crl
        └── myCA.crl
```

The command used to start the PLC is the following : 
```
docker run --rm -it -v /tmp/plc:/app/pki -p 50000:50000 -p 8080:8080 --name opcplc mcr.microsoft.com/iotedge/opc-plc:latest --pn=50000 --sph --sn=5 --sr=10 --st=uint --fn=5 --fr=1 --ft=uint --ctb --scn --lid --lsn --ref --gn=5 --ph=0.0.0.0 --csr
```

### Any log messages given by the failure

```
[08:34:49 ERR] Error while trying to read information from trusted peer store.
System.Security.Cryptography.CryptographicException: Failed to decode the X509 signature.
 ---> System.Formats.Asn1.AsnContentException: The provided data is tagged with 'Universal' class value '13', but it should have been 'Universal' class value '16'.
   at System.Formats.Asn1.AsnDecoder.CheckExpectedTag(Asn1Tag tag, Asn1Tag expectedTag, UniversalTagNumber tagNumber)
   at System.Formats.Asn1.AsnDecoder.ReadSequence(ReadOnlySpan`1 source, AsnEncodingRules ruleSet, Int32& contentOffset, Int32& contentLength, Int32& bytesConsumed, Nullable`1 expectedTag)
   at System.Formats.Asn1.AsnReader.ReadSequence(Nullable`1 expectedTag)
   at Opc.Ua.Security.Certificates.X509Signature.Decode(Byte[] crl)
   --- End of inner exception stack trace ---
   at Opc.Ua.Security.Certificates.X509Signature.Decode(Byte[] crl)
   at Opc.Ua.Security.Certificates.X509Signature..ctor(Byte[] signedBlob)
   at Opc.Ua.Security.Certificates.X509CRL.Decode(Byte[] crl)
   at Opc.Ua.Security.Certificates.X509CRL.EnsureDecoded()
   at Opc.Ua.Security.Certificates.X509CRL.get_IssuerName()
   at Opc.Ua.Security.Certificates.X509CRL.get_Issuer()
   at OpcPlc.OpcApplicationConfiguration.ShowCertificateStoreInformationAsync() in D:\a\1\s\src\OpcApplicationConfigurationSecurity.cs:line 338
```

### Expected/desired behavior

The CRL should be loaded in the trust store. 

### OS and Version?
Linux : 
```
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.6 LTS
Release:	18.04
Codename:	bionic
```

### Versions
Version : **latest, 2.5.0, 2.4.0, 2.3.0**

### Mention any other details that might be useful

The first time this error occured was during a CRL generation test in JAVA. Since we were not able to understand why it was not possible to load our CRL, we generated everything using openssl and found out it may be a bug. 

[myCA.zip](https://github.com/Azure-Samples/iot-edge-opc-plc/files/9504654/myCA.zip)

luiscantero, 2022-09-08:
@mregen could you please take a look

mregen, 2022-09-26:
see https://github.com/OPCFoundation/UA-.NETStandard/issues/1766
the CRL is generated as PEM and not as ASN.1, similar to the issue above. The UA stack expects a CRL in ASN.1 encoded format. Please try this additional conversion to create the CRL as required:
`openssl crl -inform PEM -in myca.crl.pem -outform DER -out myca.crl`

luiscantero, 2022-09-28:
@VictorRom can you please confirm if the above suggestion worked for you or if you still need help.

closed 2022-10-03 by VictorRom