Azure-Samples/iot-edge-opc-plc #175: V2.5.0 Broken

whakey, 2022-09-15:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [x] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Deploy to Azure using "Deploy to Azure" button. Can't connect using OPC UA Client. IIOT Opc UA Publisher module can't connect either.

### Expected/desired behavior
> Can connect using OPC UA client. Reverting to 2.4.2 works.

### Versions
> 2.5.0

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

gsgoodluck, 2022-09-16:
When  Plc Service Runing , how to use  MasterNodeManager  Multiple PLC

luiscantero, 2022-09-16:
@whakey I just deployed using the button and I can connect as follows:
[image]

gsgoodluck, 2022-09-16:
[image]

gsgoodluck, 2022-09-16:
When Connection Actual Device(Plc...),Check whether the device read/write request is here,
   Or how to handle actual device reads and writes

[image]

luiscantero, 2022-09-19:
@gsgoodluck could you please open a new issue and explain in more detail what you need. This issue was opened by another user for another topic, thanks.

whakey, 2022-09-19:
I tried again today, deployed completely fresh V2.5.0

Prosys OPC UA Browser, UA Expert and Azure IIOT Opc Publisher all fail to connect with BadTimeout errors, whereas all succeed with V2.4.2 (deployed using same template, just changed latest tag to 2.4.2)

Weird :D. If nobody else is having this issue then probably something is wrong on my end :(

luiscantero, 2022-09-19:
Can you please confirm that your client uses security policy etc as in my screenshot above?

closed 2022-09-26 by luiscantero

luiscantero, 2022-10-17:
@whakey could you please try the new version 2.5.1

whakey, 2022-10-18:
@luiscantero I just updated the tag from 2.4.2 to 2.5.1 and everything seems to be working!