Azure-Samples/iot-edge-opc-plc #187: OPC BadConnectionClosed Errors

terrymandin, 2022-11-10:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>
Start the opcplc simulator:
```
docker run --rm -it -p 50000:50000 -p 8080:8080 --name opcplc mcr.microsoft.com/iotedge/opc-plc:latest --pn=50000 --autoaccept --sph --sn=5 --sr=10 --st=uint --fn=20 --fr=1 --ft=uint --ctb --scn --lid --lsn --ref --gn=5 --cdn 10.2.0.5
```
Connect via IoT Edge.

Copy the pn.json text from the opcplc startup:
```
[
  {
    "EndpointUrl": "opc.tcp://99405dd7830a:50000",
    "UseSecurity": false,
    "OpcNodes": [
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/Boiler;s=Boiler" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=StepUp" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=AlternatingBoolean" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=RandomSignedInt32" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=RandomUnsignedInt32" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=51b74e55-f2e3-4a4d-b79c-bf57c76ea67c" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=1313895e-c776-4201-b893-e514864c6692" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=84a537f5-3df8-4ab0-a33c-842bdeaf6cc9" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=c9f6d2b9-c681-4e31-9684-2014c4ec860c" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=20883862-eb87-4ef7-8f08-4892d74a121a" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=DipData" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt1" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt2" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt3" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt4" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt5" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt6" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt7" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt8" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt9" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt10" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt11" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt12" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt13" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt14" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt15" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt16" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt17" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt18" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt19" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt20" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=BadFastUInt1" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=ABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWXYZABCDEFGHIJKLMNOPQRSTUVWX" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=LongString10kB" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=LongString50kB" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=LongString100kB" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=LongString200kB" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=NegativeTrendData" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=PositiveTrendData" },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=SlowUInt1", "OpcPublishingInterval": 10000 },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=SlowUInt2", "OpcPublishingInterval": 10000 },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=SlowUInt3", "OpcPublishingInterval": 10000 },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=SlowUInt4", "OpcPublishingInterval": 10000 },
      { "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=SlowUInt5", "OpcPublishingInterval": 10000 },
[... 109 lines cut ...]
Errors will begin to show in the opcplc log:
```
17:08:40 INF] PLC simulation started, press Ctrl+C to exit ...
[17:08:45 ERR] OPC: Certificate Validation failed. Reason=BadCertificateUntrusted. [CN=Micorosft.Azure.IIoT, C=DE, S=Bav, O=Microsoft, DC=DataHubEdge] [B9D119501682AE069CBFDBD1D00BE4A57593CDD0]
[... cut ...]
### Expected/desired behavior
>
Messages will show in the IoT Hub
### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)
From IoT Hub diagnostics:
```
.NET/1.36.6 (.NET 6.0.9; Linux 5.15.0-1022-azure #27~20.04.1-Ubuntu SMP Mon Oct 17 02:03:50 UTC 2022; X64) EdgeHub/1.4.2.61356014
```
### Versions
>
see above
### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

luiscantero, 2022-11-10:
Thanks, #186 fixes the issues with the Boiler and deterministic GUIDs not being picked up correctly.

luiscantero, 2022-11-11:
@mregen please see OPC UA stack errors here.