Azure-Samples/iot-edge-opc-plc #225: Certificate Push Management Inquiry

qwertyKeypad, 2023-03-14:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [x] feature request
- [x] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Calling the Push Management methods, "CreateSigningRequest" or "UpdateCertificate", the client recevies:
>> UaMethodException: status=Bad_InvalidArgument, message=One or more arguments are invalid. 
> However, going through the simulator logs I'm unable to find the exact argument that is invalid or any other information from the logs. Also, I'm not sure if the methods are implemented, could you tell me if the methods are implemented as well?

### Any log messages given by the failure
> N/A

### Expected/desired behavior
> Response or logs stating the error if any.

### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)

### Versions
>

### Mention any other details that might be useful
I'm using this https://reference.opcfoundation.org/GDS/v105/docs/7.10.7 to provide the arguments.

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

mregen, 2023-05-02:
In order to execute a certificate signing request, ensure to be authenticated with a secure session as administrator.

see here for the call: https://github.com/OPCFoundation/UA-.NETStandard/blob/29a3be5533ce4027f394995a7343c1ce91969851/Libraries/Opc.Ua.Gds.Client.Common/ServerPushConfigurationClient.cs#L568

some arguments expect null versions of the datatypes or are not supported.

luiscantero, 2023-05-15:
@qwertyKeypad do you need further help?

luiscantero, 2023-05-22:
No response

closed 2023-05-22 by luiscantero