Azure-Samples/iot-edge-opc-plc #318: certdnsnames and plchostname not used in the certificate

JeremyTheocharis, 2023-11-27:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> docker run --rm -it -p 50000:50000 -p 8080:8080 --name opcplc mcr.microsoft.com/iotedge/opc-plc:latest --pn=50000 --autoaccept --sph --sn=5 --sr=10 --st=uint --fn=5 --fr=1 --ft=uint --gn=5 --plchostname=localhost --showpnjsonph --certdnsnames=localhost

### Any log messages given by the failure
> 
[image]

### Expected/desired behavior
> localhost should show up in the DNS names in the certificate

### OS and Version?
> Docker Desktop on Windows 11

### Versions
> docker latest

### Mention any other details that might be useful
We are integrating the simulator as test simulator for the open-source project "benthos-umh" (https://github.com/united-manufacturing-hub/benthos-umh). We are troubleshooting why the server is closing the connection unexpectedly, and came up with this smaller problem. Maybe fixing this will solve the larger problem we are working on. If this does not fix our actual problem, we will probably create a separate issue for this.
> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

luiscantero, 2023-11-30:
@JeremyTheocharis any uodates?

JeremyTheocharis, 2023-11-30:
I can confirm it solved the issue, however the underlying issue is still there. But let me close this one. Thank you for your help!

closed 2023-11-30 by JeremyTheocharis

luiscantero, 2023-11-30:
Thanks for confirming. Could you please open an issue for the [OPC UA SDK](https://github.com/OPCFoundation/UA-.NETStandard/issues). Just copy-paste some info from here and post a link to this one. I will make sure that it gets triaged.