Azure-Samples/iot-edge-opc-plc #362: publishednodes.json uses Standard NodeId not Expanded NodeId

RollsChris, 2024-06-10:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [ x] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>
Run the image and generate the publishednodes.json file

### Any log messages given by the failure
>

### Expected/desired behavior
>
Generated publishednodes.json uses standard NodeId not Expanded NodeId.

### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)

### Versions
>

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

luiscantero, 2024-06-14:
@RollsChris the pn.json output uses expanded NodeId, e.g. `{ "Id": "nsu=http://microsoft.com/Opc/OpcPlc/Boiler;i=6211" }`

RollsChris, 2024-06-14:
Yes but I would like away to use standard node IDs? Is that possible

luiscantero, 2024-06-14:
Ok, from the title and description it sounded like right now we use standard nodeid and you would like to use expanded instead. Currently we don't support this. This file is generated for the Publisher, which understands this format. What is your use-case?

luiscantero, 2024-06-18:
No response.

closed 2024-06-18 by luiscantero