Azure-Samples/iot-edge-opc-plc #366: Subscription feature in K8S.

chunchiehdev, 2024-06-18:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [x] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```
I noticed that when I deploy `opcplc` to AKS and set the replica to 3, it will create three pods.

However, when I subscribe to `opcplc` from client A with ns=1,s='temperature', value=25, there might be a log in pod1 indicating that someone has subscribed. Then, if I change the value to 30 for ns=1,s='temperature' from client B (I forward the pod2 to my localhost 4840), it might result in an "Error publishing." Also client A won't change the value. 

I suspect the reason is that I started 3 pods. When I set replicas to 1, everything works fine.

### Minimal steps to reproduce
> Deploy `opc-plc` to AKS with replicas set to 3 -> Use client A to subscribe to a node, then use client B to modify the node -> After that, an error occurs and client A cannot read the updated node value. 

### Any log messages given by the failure
>

```
<6>2024-06-18T10:27:48.259Z [opcua@311 ChannelId:"3"] - Session CREATED, Id=ns=7;i=947326284, Name=Session i=0, ChannelId=0f6aa8ae-8f09-45aa-882e-264691a9510b-3, User=Anonymous
<6>2024-06-18T10:27:48.259Z [opcua@311 ChannelId:"3"] - Server - SESSION CREATED. SessionId=ns=7;i=947326284
<6>2024-06-18T10:27:48.260Z [opcua@311 ChannelId:"3"] - Session VALIDATED, Id=ns=7;i=947326284, Name=Session i=0, ChannelId=0f6aa8ae-8f09-45aa-882e-264691a9510b-3, User=Anonymous
<6>2024-06-18T10:27:48.260Z [opcua@311 ChannelId:"3"] - Session FIRST ACTIVATION, Id=ns=7;i=947326284, Name=Session i=0, ChannelId=0f6aa8ae-8f09-45aa-882e-264691a9510b-3, User=Anonymous
<6>2024-06-18T10:27:48.260Z [opcua@311 ChannelId:"3"] - Server - SESSION ACTIVATED.
<6>2024-06-18T10:27:48.288Z [opcua@311 ChannelId:"3"] - Subscription CREATED, SessionId=ns=7;i=947326284, Id=2832759200, Priority=0, Publishing=500, KeepAlive=10, LifeTime=7200, MaxNotifications=2000, Enabled=True
<6>2024-06-18T10:27:48.294Z [opcua@311 ChannelId:"3"] - Subscription ITEMS CREATED, Id=2832759200, ItemCount=1, ItemsToCheck=1, ItemsToPublish=0
<6>2024-06-18T10:27:52.922Z [opcua@311 ChannelId:"2"] - Subscription DELETED, SessionId=(null), Id=2832759199, SeqNo=2, MessageCount=0
<6>2024-06-18T10:27:52.927Z [opcua@311 ChannelId:"2"] - Subscription ITEMS DELETED, Id=2832759199, ItemCount=0, ItemsToCheck=0, ItemsToPublish=0
<6>2024-06-18T10:27:52.928Z [opcua@311 ChannelId:"2"] - Session CLOSED, Id=ns=7;i=947326196, Name=Session i=0, ChannelId=0f6aa8ae-8f09-45aa-882e-264691a9510b-2, User=Anonymous
<6>2024-06-18T10:27:52.929Z - TCPSERVERCHANNEL ProcessCloseSecureChannelRequest success, ChannelId=2, TokenId=2, Socket=00CF68CC
<6>2024-06-18T10:27:52.930Z - ChannelId 2: closed
<3>2024-06-18T10:27:52.923Z [opcua@311 ChannelId:"2"] - Error publishing
Opc.Ua.ServiceResultException: BadSessionClosed
   at Opc.Ua.Server.StandardServer.Publish(RequestHeader requestHeader, SubscriptionAcknowledgementCollection subscriptionAcknowledgements, UInt32& subscriptionId, UInt32Collection& availableSequenceNumbers, Boolean& moreNotifications, NotificationMessage& notificationMessage, StatusCodeCollection& results, DiagnosticInfoCollection& diagnosticInfos)
   at OpcPlc.PlcServer.Publish(RequestHeader requestHeader, SubscriptionAcknowledgementCollection subscriptionAcknowledgements, UInt32& subscriptionId, UInt32Collection& availableSequenceNumbers, Boolean& moreNotifications, NotificationMessage& notificationMessage, StatusCodeCollection& results, DiagnosticInfoCollection& diagnosticInfos) in D:\a\1\s\src\PlcServer.cs:line 172
<3>2024-06-18T10:27:52.923Z [opcua@311 ChannelId:"2"] - Error publishing
Opc.Ua.ServiceResultException: BadSessionClosed
```

### Expected/desired behavior
> Is it possible for all three pods to update the node value?

### OS and Version?
> Linux

### Versions
> 22.04.3 

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

luiscantero, 2024-06-18:
Can you please elaborate your use-case and explain the reason you want/need to run multiple replicas?
At first sight, this doesn't look like a supported scenario.

chunchiehdev, 2024-06-18:
Thank you for your response.

By running multiple replicas, I aim to ensure that the OPC PLC service remains available even if one or more Pods fail.
Also distribute the client requests across different Pods to enhance performance.

Are there any best practices or solutions for synchronizing node values across multiple replicas in such a setup?

cristipogacean, 2024-06-18:
@chunchiehdev, the opc plc simulation opc ua server is designed to act like a standalone containerized application. It is not designed to run in a k8s cluster as a distributed architecture. If you spin multiple instances just do this using separate deployments with replica set always to 1. Increasing replica setting to more than 1 will just end up in un-predictable behavior of the multiple instances of the opc plc. They are completely unaware about any peers deployed in the same cluster and act as independent standalone applications.

chunchiehdev, 2024-06-18:
Thank you for the detailed explanation. Now I understand.

closed 2024-06-18 by chunchiehdev