Azure-Samples/iot-edge-opc-plc #370: Alarm Confirm/acknowledge results in Status Code BAD

DerGary, 2024-06-21:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Start the Application with Alarms that are raised automatically
> Connect to the OPCUA Server via Ua Expert for example
> Confirm or acknowledge an alarm
> Result will be:
[image]

### Any log messages given by the failure
> none

### Expected/desired behavior
> The result is Good

### OS and Version?
> Executed in the docker container image `mcr.microsoft.com/iotedge/opc-plc:latest`

### Mention any other details that might be useful

The error may be located in the opc ua library.

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

luiscantero, 2024-06-28:
Thanks for reporting this, we are investigating and will get back ASAP.

luiscantero, 2025-10-23:
There seems to be an issue with the stack that we are unable to fix at the moment.