Azure-Samples/iot-edge-opc-plc #395: Server doesn't close sessions after client closes the session (BadTooManySessions)

DvdKhl, 2024-09-21:
### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Basically start the container as described in the readme and open/close a connection 101 times

Detailed:
 - Start the OpcPlc Container with
   - `docker run --rm -it -p 50000:50000 -p 8080:8080 --name opcplc mcr.microsoft.com/iotedge/opc-plc:latest --pn=50000 --autoaccept --sph --sn=5 --sr=10 --st=uint --fn=5 --fr=1 --ft=uint --gn=5`
 - Download [IoTEdgeOpcPlcTooManySessions.zip](https://github.com/user-attachments/files/17083715/IoTEdgeOpcPlcTooManySessions.zip)
   - Open and run Application
   -  Basically just copied parts of the [reference opc client](https://github.com/OPCFoundation/UA-.NETStandard/blob/master/Applications/ConsoleReferenceClient/Program.cs) code
   - The issue can also be reproduced by just using uaexport and connecting/disconnecting the session 101 times

After this, the server basically becomes unusable and has to be restarted.

### Any log messages given by the failure
> BadTooManySessions

Client Side:  
```
Connecting to... opc.tcp://localhost:50000
BadCertificateUntrusted 'Certificate is not trusted.'
Untrusted Certificate accepted. Subject = CN=OpcPlc
BadCertificateUntrusted 'Certificate is not trusted.'
Untrusted Certificate accepted. Subject = CN=OpcPlc
New Session Created with SessionName = Quickstart Console Reference Client
Disconnecting...
Session Disconnected.
Connected and disconnected successfully 98 times

Connecting to... opc.tcp://localhost:50000
BadCertificateUntrusted 'Certificate is not trusted.'
Untrusted Certificate accepted. Subject = CN=OpcPlc
BadCertificateUntrusted 'Certificate is not trusted.'
Untrusted Certificate accepted. Subject = CN=OpcPlc
New Session Created with SessionName = Quickstart Console Reference Client
Disconnecting...
Session Disconnected.
Connected and disconnected successfully 99 times

Connecting to... opc.tcp://localhost:50000
BadCertificateUntrusted 'Certificate is not trusted.'
Untrusted Certificate accepted. Subject = CN=OpcPlc
BadCertificateUntrusted 'Certificate is not trusted.'
Untrusted Certificate accepted. Subject = CN=OpcPlc
Create Session Error : BadTooManySessions
```

ServerSide:
```
<6>2024-09-21T10:18:19.116Z - Sender Certificate: [CN=Console Reference Client, C=US, S=Arizona, O=OPC Foundation, DC=hestia] [7B67E97E74B1B49C3886854DE16D5A37BDAC76BA]
<6>2024-09-21T10:18:19.116Z - ChannelId 303: Token #0 created. CreatedAt=10:18:19.116. Lifetime=3600000.
<6>2024-09-21T10:18:19.116Z - ChannelId 303: Token #1 activated. CreatedAt=10:18:19.116. Lifetime=3600000.
<4>2024-09-21T10:18:19.123Z [opcua@311 ChannelId:"303"] - Trusting certificate CN=Console Reference Client, C=US, S=Arizona, O=OPC Foundation, DC=hestia because of corresponding command line option
<4>2024-09-21T10:18:19.123Z [opcua@311 ChannelId:"303"] - Validation errors suppressed:  [CN=Console Reference Client, C=US, S=Arizona, O=OPC Foundation, DC=hestia] [7B67E97E74B1B49C3886854DE16D5A37BDAC76BA]
<3>2024-09-21T10:18:19.124Z [opcua@311 ChannelId:"303"] - Server - SESSION CREATE failed. BadTooManySessions
<3>2024-09-21T10:18:19.127Z [opcua@311 ChannelId:"303"] - CreateSession error
Opc.Ua.ServiceResultException: BadTooManySessions
   at Opc.Ua.Server.StandardServer.CreateSession(RequestHeader requestHeader, ApplicationDescription clientDescription, String serverUri, String endpointUrl, String sessionName, Byte[] clientNonce, Byte[] clientCertificate, Double requestedSessionTimeout, UInt32 maxResponseMessageSize, NodeId& sessionId, NodeId& authenticationToken, Double& revisedSessionTimeout, Byte[]& serverNonce, Byte[]& serverCertificate, EndpointDescriptionCollection& serverEndpoints, SignedSoftwareCertificateCollection& serverSoftwareCertificates, SignatureData& serverSignature, UInt32& maxRequestMessageSize)
   at OpcPlc.PlcServer.CreateSession(RequestHeader requestHeader, ApplicationDescription clientDescription, String serverUri, String endpointUrl, String sessionName, Byte[] clientNonce, Byte[] clientCertificate, Double requestedSessionTimeout, UInt32 maxResponseMessageSize, NodeId& sessionId, NodeId& authenticationToken, Double& revisedSessionTimeout, Byte[]& serverNonce, Byte[]& serverCertificate, EndpointDescriptionCollection& serverEndpoints, SignedSoftwareCertificateCollection& serverSoftwareCertificates, SignatureData& serverSignature, UInt32& maxRequestMessageSize) in D:\a\1\s\src\PlcServer.cs:line 148
<4>2024-09-21T10:18:19.132Z [opcua@311 ChannelId:"303"] - SERVER - Service Fault Occurred. Reason=BadTooManySessions
<6>2024-09-21T10:18:19.134Z - TCPSERVERCHANNEL ProcessCloseSecureChannelRequest success, ChannelId=303, TokenId=1, Socket=02140105
<6>2024-09-21T10:18:19.134Z - ChannelId 303: closed
```

### Expected/desired behavior
> Properly close the session after the client closes the session

### OS and Version?
> Windows 11 23H2

### Versions
NET 8.0
opc-plc:latest (21-09-2024)
OPCFoundation.NetStandard.Opc.Ua.Client 1.5.374.118

DvdKhl, 2025-05-04:
This seems to be fixed? At least I couldn't reproduce it on .NET9 with 1.5.376.213.

mregen, 2025-05-06:
yes, this is fixed

closed 2025-05-06 by mregen