Azure-Samples/iot-edge-opc-plc #400: .uanodes file loading limit?

Fabolo64, 2024-11-04:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [ ] feature request
- [x] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```
This, in reality, is a request of info about possible configuration limit in loading a set of nodes to be used in a custom simulation.

### Minimal steps to reproduce
>
create a class implementing *IPluginNodes*
load uanodes file in *LoadPredefinedNodes* method
*AddPredefinedNode* to _plcNodeManager in *AddToAddressSpace* method

my nodes, when in limited number, do appear under my custom folder when browsing from an OPC-UA Client, but when I start to add all needed ones and recompiles with bigger .uanodes file (I create them with UA-ModelCompiler ) it seems they hit a limit and then simply don't show.

### Any log messages given by the failure
>
no errors/warning.

tried to increment values in configuration without success: 
```
var operationLimits = new OperationLimits() {
            MaxMonitoredItemsPerCall = 25000,
            MaxNodesPerBrowse = 25000,
            MaxNodesPerHistoryReadData = 10000,
            MaxNodesPerHistoryReadEvents = 10000,
            MaxNodesPerHistoryUpdateData = 10000,
            MaxNodesPerHistoryUpdateEvents = 10000,
            MaxNodesPerMethodCall = 10000,
            MaxNodesPerNodeManagement = 10000,
            MaxNodesPerRead = 25000,
            MaxNodesPerWrite = 10000,
            MaxNodesPerRegisterNodes = 10000,
            MaxNodesPerTranslateBrowsePathsToNodeIds = 10000,
        };
```

### Expected/desired behavior
>

### OS and Version?
> 

Windows 10

### Versions
>
OpcPlc v2.12.23

luiscantero, 2024-11-19:
@Fabolo64 we discussed this and are not aware of loading limits for the .uanodes files.

luiscantero, 2024-11-28:
We would need a way to repro this to investigate futher.

closed 2024-11-28 by luiscantero