Azure-Samples/iot-edge-opc-plc #403: SourceTimestamp does not update correctly for CurrentTime on opcplc

erhardeibl, 2024-11-29:
### This issue is for a: (mark with an `x`)
```
- [ x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
Subscribe to the node Root\Objects\Server\ServerStatus\CurrentTime (nodeID 2258) on opcplc. 
According to the [spec](https://reference.opcfoundation.org/Core/Part4/v105/docs/7.11.3) "The [sourceTimestamp](https://reference.opcfoundation.org/search/357?t=sourceTimestamp) [...] should indicate the time of the last change of the [value](https://reference.opcfoundation.org/search/357?t=value) or [statusCode](https://reference.opcfoundation.org/search/357?t=statusCode). The [sourceTimestamp](https://reference.opcfoundation.org/search/357?t=sourceTimestamp) should be generated as close as possible to the source of the value but the timestamp needs to be set always by the same physical clock."
But it does not update when the value changes. When you subscribe to the topic you will receive the same SourceTimestamp over and over again while the value changes as expected.

The first subscription is Kepserver, the second opcplc. Notice that Source Timestamp for opcplc is very different from all other timestamps:
[image]

Message diff in MQTT Explorer - SourceTimestamp is identical:
[image]

### Expected/desired behavior
The Source Timestamp should update when the value updates as defined by the spec.

### OS and Version?
Windows 11 Version 10.0.26100 Build 26100

### Versions
NET 8.0.404
OpcPlc v2.12.24.0
OPC UA SDK 1.5.374.125 from 10/31/2024 17:37:56

luiscantero, 2024-12-06:
@erhardeibl sees like a stack issue as the OPC PLC does not control this value. Martin is looking into it.

mregen, 2024-12-06:
Its a bug in the OPC UA server library, will be fixed in the next release

luiscantero, 2025-01-24:
Fixed by stack update #408

closed 2025-01-24 by luiscantero